ros-planning/navigation #915: Costmap_2d for 2-D quadrotors navigation

IhabMohamed, 2019-08-25:
Hi everyone,

I have an MPC controller for performing 2d navigation on obstacles environments (like cylinder-like environment) which is run on [Rotors Simulator.](https://github.com/ethz-asl/rotors_simulator). Right now, the controller listens to  ``/gazebo/model_states`` in order to get the map of obstacles located in the world. This map is a 2D grid, where "0" refers to free-space while "1" refers to obstacles. I would like to use ``costmap_2d`` instead based on depth sensor (camera). 

My question is how can I use "costmap_2d" package properly for that purpose, without using other packages mentioned here?

Many thanks in advance!
Regards,
Ihab Mohamed

DLu, 2019-08-30:
Please post open ended questions like this on answers.ros.org. The `costmap_2d` package most readily accepts `PointCloud2` and `LaserScan` messages.

closed 2019-08-30 by DLu

IhabMohamed, 2019-08-30:
Thanks, @DLu :-)