ros-planning/navigation #917: global_planner A* NO PATH

alliansmover, 2019-08-26:
I often make mistakes when I use A* planning.

NO PATH!
[ERROR] [1566820175.641018822, 322.374000000]: Failed to get a plan from potential when a legal potential was found. This shouldn't happen.

Can you tell me what is the problem?
thanks

anish0x2a, 2019-10-08:
Try setting `use_grid_path` to false.

bxc237, 2019-11-14:
> Try setting `use_grid_path` to false.
If I recall correctly,  use_grid_path should be set to true in order to make A* work, although I don't know why... If someone can explain the reason behind it, that will be great.

willzoe, 2020-05-07:
I also encountered the same problem, even if set `use_grid_path` to `false`. 

I posted my question [here](https://answers.ros.org/question/350510/some-errors-when-using-globalplanner/), I hope to get help.

DLu, 2020-07-03:
Which version of global_planner are you using?

Ellinier, 2020-08-31:
I also encountered the same problem when using A* planning and grid path.

I found it happened when the start point was (0.0, 0.0) or the end point was (0.0, 0.0). The same problem was also occurred in other situations such as the start point was (0.0, 0.1) and the end point was (0.5, 0.0).

I looked into the code and found it may mostly due to the bugs in grid_path.cpp 
At least in the Line70-71
`if (min_x == 0 && min_y == 0)
            return false;`
when the start point was (0.0, 0.0), even if it got a path from the end point to the start point, it returned false.

I think there is also something wrong with the getPath mechanism, especially when dealing with some corner start points or end points, but I don't know how to fix it.

Can anyone help to solve it?

Many thanks! @DLu

Astaxuqichao, 2024-04-30:
I encounter the same issue.
I find the potential map is incorrect.
the neutral_cost almost do not work, finding that  the cost in _calculatePotential_ is is limited to 255 , cause of limitations of char
I fix the calculatePotential function, like this.
`virtual float calculatePotential(float* potential, float cost, int n, float prev_potential=-1)`

It did work, hoping  that help you.

RookieAllOut, 2025-01-04:
> I encounter the same issue. I find the potential map is incorrect. the neutral_cost almost do not work, finding that the cost in _calculatePotential_ is is limited to 255 , cause of limitations of [...]

This fixed the NO PATH! … error for me.

closed 2025-06-04 by mikeferguson