ros-planning/navigation #920: Why is uniformPoseGenerator() a static method in AMCL?

mkhansen-intel, 2019-08-30:
https://github.com/ros-planning/navigation/blob/970eefe147c4f168281b3ca3e378a81baf350300/amcl/src/amcl_node.cpp#L158

I was trying to clean up AMCL a little bit in navigation2 and saw this. I can't find anywhere it's used outside of AMCL. I wanted to make it a regular method. Only reason I can think of it's static is if there were multiple instances of AMCL in the same process and you wanted them to share a single pose generator?

Does anyone know? Looking at the git blame on that line doesn't really tell me as it's just "migration part 1" and its from 2010.

SteveMacenski, 2019-08-31:
@mkhansen-intel I think its static so it has access to this static member https://github.com/ros-planning/navigation/blob/970eefe147c4f168281b3ca3e378a81baf350300/amcl/src/amcl_node.cpp#L1022

Which is static for... threading reasons?

DLu, 2019-09-03:
[It's been that way since the beginning of time, aka 2009](https://github.com/ros-planning/navigation/commit/1f335323a605b49b4108a845c55a7c1ba93a6f2e#diff-ef9bfe45efbfe60a5512d679a78bceb9R105). Beyond that I don't know.

B0gdar, 2019-09-05:
I think this refactoring task has been even a thing since the begging of time, AMCL is a 15-19 years old project that have lived two lifes already (before ROS (C ->C++) and with ROS).

Half a year ago, after reviewing the amcl source code I got interested about some legacy code present in amcl/src/amcl/pf, then pull the thread and arrived to the original reference source of the ROS amcl version: [the player project a.ka One Hell Of a Robot, Gerkey and Howard; early 2000's till 2011](http://playerstage.sourceforge.net/). The player project was bonded first with the stage simulator and then it is mentioned at some point of the project some references to ROS and early versions of gazebo (2011 aprox). 

The functions you mentioned are heavily related to the modified particle filter feature folder. At player project it is mentioned in different parts the following header (p.e. in the gridmap_init factory creation):
```c++
// A factory creation function, declared outside of the class so that it
// can be invoked without any object context (alternatively, you can
// declare it static in the class).  In this function, we create and return
// (as a generic Driver*) a pointer to a new instance of this driver.
Driver* gridmap_Init(ConfigFile* cf, int section)
{
  // Create and return a new instance of this driver
  return ((Driver*) (new gridmap(cf, section)));
}
```
So it has been an architectural decision made 7 years ago (?) at ROS Electric times, most likely based on some workflow already present on player project... I have not gone deeply on this specific topic, if I find something interesting I may update.

mikeferguson, 2020-03-03:
Going to mark this as a question and close, as I don't think it needs to be addressed in ROS1

closed 2020-03-03 by mikeferguson