ros-planning/navigation #922: Path is made very close to obstacles

fayyazpocker, 2019-09-09:
Global planner makes the path very close to the obstacles as a result of which the robot hits the obstacles. Following are the configuration used for move_base.

Global planner: global_planner/GlobalPlanner

**costmap_common_param**

obstacle_range: 8
raytrace_range: 15
 footprint: [[-0.9, -0.7], [-0.9, 0.7], [0.9, 0.7], [0.9, -0.7]]
inflation_radius: 1
....

**global_costmap_param**

Global planner:
 lethal cost: 253
 neutral cost: 66
 cost_factor: 0.55

global_costmap:
  global_frame: map 
  robot_base_frame: base_link 
  update_frequency: 3.0
  static_map: true

[image]

Increasing inflation radius is one probable solution but because of that every region gets affected which i don't want.
I tried increasing footprint expecting that the global planner will make the path away from the obstacle. However, i could not find much change either.

I tried playing around with lethal cost, neutral cost and cost_factor but still i am not able to get the path accordingly.

DLu, 2019-09-09:
A) How big is your robot? 
B) What is your [inscribed_radius](http://wiki.ros.org/costmap_2d#Inflation) set to?

fayyazpocker, 2019-09-10:
The robot is of rectangle shape. It has a length of 1.5m and a width of 1.12m. I have added an extra footprint as shown in the figure in case of safety.

[image]

I have not set inscribed_radius manually as i thought given the footprint, the inscribed and circumscribed radius is calculated automatically. However, following a [query](https://answers.ros.org/question/62277/inscribed-and-circumscribed-radii-of-the-robot-used-despite-footprint-being-set/) in discourse, inflation radius was also set above 1.14 (sqrt(0.9^2 + 0.7^2)). I am not able to get better result despite of that as well.

DLu, 2019-09-10:
Since this is more of a configuration problem than an actual bug, please open your own post on [answers.ros.org](answers.ros.org) so that we can iterate a bit more. Please include your full costmap settings, including the resolution.

closed 2019-09-10 by DLu