ros-planning/navigation #927: Robot moves slowly when goal is in the negative "odom" frame direction

davidteixeira, 2019-10-02:
This issue is related to #503. Whenever the robot moves in the negative x direction of the odom frame, it moves very slowly, never reaching the maximum speed defined in the DWA local planner parameters. The particularity here is that the robot should only move forward (min_vel_x = -0.05, max_vel_x = 1.0).
What I found regarding the velocity space is the following:

- min_vel_x = -1.0 and max_vel_x = 1.0:
velocity space = [-0.2, 0.2] for vel = 0
velocity space = [0.8, 1.0] for vel = 1
velocity space = [-1.0, -0.8] for vel = -1

- min_vel_x = -0.05 and max_vel_x = 1.0:
velocity space = [-0.05, 0.2] for vel = 0
velocity space = [0.8, 1.0] for vel = 1
velocity space = [-0.05, -0.8] for vel = -1

So it seems the velocity space is not adequately created for the robot in this particular case. I might be mixing up frames of reference here, not sure. In any case, I changed [this line](https://github.com/ros-planning/navigation/blob/e5162e66ca7fbe2ee274f81eb98b1c0b2868767d/base_local_planner/src/simple_trajectory_generator.cpp#L108) to `max_vel[0] = std::min(max_vel_x, std::abs(vel[0]) + acc_lim[0] * sim_period_);` and the problem went away.

Now, this works for my specific case, but I would like to know if it makes sense for someone more familiar with the code and if it would be worth implementing properly (as I believe this breaks the velocity space definition for robots that move in both directions).

davidteixeira, 2019-10-04:
Alright, I found the problem: it was my odometry calculation.
Specifically (if anyone runs into this issue), the velocity was being published in the 'base_link' frame, but calculated in the 'odom' frame, i.e., for this robot, which is non-holonomic, the velocity of the robot should be the same as v<sub>x</sub>; v<sub>y</sub> should always equal 0. I got the [idea](https://github.com/jfstepha/differential-drive/blob/5ef3d6d1dea71e8856e73e578c9da1ed0ea08119/scripts/diff_tf.py#L147) from the differential-drive package.

closed 2019-10-04 by davidteixeira

kishuinvain1, 2019-10-11:
> Alright, I found the problem: it was my odometry calculation.
> Specifically (if anyone runs into this issue), the velocity was being published in the 'base_link' frame, but calculated in the 'odom' frame, i.e., for this robot, which is non-holonomic, the velocity of the robot should be the same as vx; vy should always equal 0. I got the [idea](https://github.com/jfstepha/differential-drive/blob/5ef3d6d1dea71e8856e73e578c9da1ed0ea08119/scripts/diff_tf.py#L147) from the differential-drive package.

I've been facing the same problem.
Can you please elaborate it.

davidteixeira, 2019-10-20:
How are you calculating the velocity of the robot? Can you share some code?