ros-planning/navigation #942: Voxel costmap plugin adding incorrect obstacles at ground level

VorpalBlade, 2019-11-13:
The voxel map doesn't clear properly at the ground level when using a Velodyne VLP-16 3D laser as the input sensor.

Specifically the problem is the code segment (in `VoxelLayer::updateBounds`):

```c++
if (cloud.points[i].z < origin_z_)
{
if (!worldToMap3D(cloud.points[i].x, cloud.points[i].y, origin_z_, mx, my, mz))
  continue;
}
else if (!worldToMap3D(cloud.points[i].x, cloud.points[i].y, cloud.points[i].z, mx, my, mz))
{
continue;
}
```

This means that if the ray passes below the ground level `origin_z_` (or exactly at) it will be treated as an obstacle at the ground level. This means that if the ground can be seen at all, there will be obstacles at it. These obstacles will then lead to obstacles in the cost map.

Since my visibility of the ground is pretty sparse (a few "rings" that the laser sees) I cannot really compensate for this with the `mark_threshold` parameter, that could be possible if you had a dense view of the floor from for example a 3D camera.

Another option could be to restrict the beams using `min_obstacle_height`, but then the beams are discarded entirely and will not help with clearing obstacles, so that is not an option either for me.

To work as I would expect the first if statement should simply be removed. However, this looks intentionally done, so I would like to know the use case for it. As a compromise I suggest adding a parameter to disable this case (as I did in a patch against kinetic in commit 4b9ec9b4b94f05b22d2cfd1f5cb604973b09412e). Presumably this would need to be ported to a newer version (e.g. melodic), which I could do if there is interest in it.

ROS version: Kinetic

SteveMacenski, 2019-11-13:
> Another option could be to restrict the beams using min_obstacle_height, but then the beams are discarded entirely and will not help with clearing obstacles, so that is not an option either for me.

You can have a given observation source be only for marking and another of the same topic only for clearing separately. That will give you the functionality you want so you can clear to the ground but mark only to a given minimum height. 

I don't think it should be removed, but I see why its irritating to you. Another option is to have a pre-processing stage where you remove the ground reading points, its pretty typical for users to filter or otherwise process sensor data before being placed into an environmental model.

VorpalBlade, 2019-11-13:
> You can have a given observation source be only for marking and another of the same topic only for clearing separately. That will give you the functionality you want so you can clear to the ground [...]

That could probably work yes. I'm afraid I'm in a bit of a time crunch on the project I'm working on, so I will go for the patch I already tested instead of this though.

> I don't think it should be removed, but I see why its irritating to you. Another option is to have a pre-processing stage where you remove the ground reading points, its pretty typical for users to filter or otherwise process sensor data before being placed into an environmental model.

That would also remove clearing along the beams towards the ground. Unless again separate clearing and marking sources were used (with only the marking one being pre-processed)

SteveMacenski, 2019-11-13:
> That could probably work yes. I'm afraid I'm in a bit of a time crunch on the project I'm working on, so I will go for the patch I already tested instead of this though.

got it, just letting you know you have options in this architecture to do what you propose. I think with that this ticket could be closed unless you think there's an action?

VorpalBlade, 2019-11-13:
That seems fair enough.

closed 2019-11-13 by VorpalBlade