ros-planning/navigation #944: ROS Noetic support?

AutonomousHansen, 2019-11-27:
In going through the [Python3 transition tutorial](http://wiki.ros.org/UsingPython3) I found that costmap_2d has Python 2 dependencies, and has no listed entry on the [yaml where they document Noetic readiness](https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml).

Are there plans to support Noetic?

mikeferguson, 2019-11-27:
We're pretty high on the dependency list, we already added the msgs to Noetic, but it'll probably be a little while before this is ready to release to Noetic. AFAIK, the only Python in these packages are dynamic reconfigure and maybe some tests.

sloretz, 2020-04-01:
It looks like the `navigation` repo is unblocked now that `diagnostics` has been released into ROS Noetic.

mikeferguson, 2020-04-01:
noetic-devel branch created. going to do a release of our last melodic release into noetic before I start merging the other "hold for noetic" PRs

mikeferguson, 2020-04-01:
@sloretz - it looks like we're still blocked by orocos_kinematics_dynamics

sloretz, 2020-04-01:
> it looks like we're still blocked by orocos_kinematics_dynamics

For Noetic the plan is to use the upstream Debian packages instead of ROS packages since the upstream versions are also the latest versions http://wiki.ros.org/noetic/Migration#Use_Orocos_KDL_and_BFL_rosdep_keys

mikeferguson, 2020-04-01:
Ah, ok. I've opened a PR to get pull request building going, since I don't have a 20.04 machine to test on: https://github.com/ros/rosdistro/pull/24319

mikeferguson, 2020-04-02:
Status update:

 * updated to use rosdep keys
 * patched some python3 issues - we initially had 9 failing tests, now down to 1
 * still working on that last test (this test has been flaky in the past, appears to not work at all now). I've spent some time with Valgrind on it tonight - have pushed one potential patch, have another that I will try out in the morning if the first doesn't fix it.

Should have a first release tomorrow.

mikeferguson, 2020-04-02:
First release: https://github.com/ros/rosdistro/pull/24321 - there will still be some API changes shortly.

closed 2020-04-02 by mikeferguson