ros-planning/navigation #951: /move_base_/global_costmap/costmap could not change with map used command "rosrun map_server map_server map.yaml"

shanpenghui, 2019-12-25:
After update the ros and teb, it do not work. When i start navigation stack at the beginning, global_costmap is matched with the static map. But after change map with map_server, global_costmap is not matched with the static map. What is the problem?See the pictures bellow.
[image]

[image]

imutkarshpatil, 2019-12-27:
Hi @shanpenghui,

If you are using RViz for visualization purpose, try unchecking and checking the checkbox for display of global_costmap in the display tab, after you update the map with map_server. This will cause RViz to make new request to fetch an updated global_costmap.

mikeferguson, 2020-03-03:
Pretty sure this is another case of #959 - which should be fixed by https://github.com/ros-planning/navigation/commit/37a979f7bbf3316196fe95646b6d75f9a79fd135

closed 2020-03-03 by mikeferguson