ros-planning/navigation #960: Error during installation on Raspberry Pi with ROS Kinetic due to YAML

antoninovara, 2020-01-22:
Hello everybody!

I'm installing ROS Kinetic in Raspberry Pi 3 b+ following this tutorial.
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

My software version of Raspbian is this.
https://photos.app.goo.gl/PDkqA5t4tihD6RV5A

When I try to install "map_server" I get this error.
https://photos.app.goo.gl/rq4LSG4zSKA8sTU36

And this is a video with the complete proccess.
https://youtu.be/WxnyF_9ugiw

I don't know how to solve the problem. Can someone help me?

Thank you very much and regards

SteveMacenski, 2020-01-22:
Please post the actual compiler error. Images (or links to images) are not searchable.

antoninovara, 2020-01-22:
> Hello everybody!

I'm installing ROS Kinetic on Raspberry Pi 3 b+ following this tutorial. http://wiki.ros.org/ROSberryPi/Instal...

My software version of Raspbian is this. https://photos.app.goo.gl/PDkqA5t4tih...

When I try to install "map_server" I get this error.

/usr/include/yaml-cpp/node/detail/node.h:113:39: error: ‘shared_memory_holder’ has not been declared
   void insert(node& key, node& value, shared_memory_holder pMemory) {
                                       ^
/usr/include/yaml-cpp/node/detail/node.h:121:29: error: ‘shared_memory_holder’ has not been declared
   node* get(const Key& key, shared_memory_holder pMemory) const {
                             ^
/usr/include/yaml-cpp/node/detail/node.h:128:29: error: ‘shared_memory_holder’ has not been declared
   node& get(const Key& key, shared_memory_holder pMemory) {
                             ^
/usr/include/yaml-cpp/node/detail/node.h:134:31: error: ‘shared_memory_holder’ has not been declared
   bool remove(const Key& key, shared_memory_holder pMemory) {
                               ^
/usr/include/yaml-cpp/node/detail/node.h:138:24: error: ‘shared_memory_holder’ has not been declared
   node* get(node& key, shared_memory_holder pMemory) const {
                        ^
/usr/include/yaml-cpp/node/detail/node.h:144:24: error: ‘shared_memory_holder’ has not been declared
   node& get(node& key, shared_memory_holder pMemory) {
                        ^
/usr/include/yaml-cpp/node/detail/node.h:150:26: error: ‘shared_memory_holder’ has not been declared
   bool remove(node& key, shared_memory_holder pMemory) {
                          ^
/usr/include/yaml-cpp/node/detail/node.h:157:21: error: ‘shared_memory_holder’ has not been declared
                     shared_memory_holder pMemory) {
                     ^
/usr/include/yaml-cpp/node/detail/node.h:162:3: error: ‘shared_node_ref’ does not name a type
   shared_node_ref m_pRef;
   ^
/usr/include/yaml-cpp/node/detail/node.h: In constructor ‘YAML::detail::node::node()’:
/usr/include/yaml-cpp/node/detail/node.h:21:12: error: class ‘YAML::detail::node’ does not have any field named ‘m_pRef’
   node() : m_pRef(new node_ref) {}
            ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘bool YAML::detail::node::is(const YAML::detail::node&) const’:
/usr/include/yaml-cpp/node/detail/node.h:25:43: error: ‘m_pRef’ was not declared in this scope
   bool is(const node& rhs) const { return m_pRef == rhs.m_pRef; }
                                           ^
/usr/include/yaml-cpp/node/detail/node.h:25:57: error: ‘const class YAML::detail::node’ has no member named ‘m_pRef’
   bool is(const node& rhs) const { return m_pRef == rhs.m_pRef; }
                                                         ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘const YAML::detail::node_ref* YAML::detail::node::ref() const’:
/usr/include/yaml-cpp/node/detail/node.h:26:40: error: ‘m_pRef’ was not declared in this scope
   const node_ref* ref() const { return m_pRef.get(); }
                                        ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘bool YAML::detail::node::is_defined() const’:
/usr/include/yaml-cpp/node/detail/node.h:28:36: error: ‘m_pRef’ was not declared in this scope
   bool is_defined() const { return m_pRef->is_defined(); }
                                    ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘const YAML::Mark& YAML::detail::node::mark() const’:
/usr/include/yaml-cpp/node/detail/node.h:29:37: error: ‘m_pRef’ was not declared in this scope
   const Mark& mark() const { return m_pRef->mark(); }
                                     ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘YAML::NodeType::value YAML::detail::node::type() const’:
/usr/include/yaml-cpp/node/detail/node.h:30:41: error: ‘m_pRef’ was not declared in this scope
   NodeType::value type() const { return m_pRef->type(); }
                                         ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘const string& YAML::detail::node::scalar() const’:
/usr/include/yaml-cpp/node/detail/node.h:32:46: error: ‘m_pRef’ was not declared in this scope
   const std::string& scalar() const { return m_pRef->scalar(); }
                                              ^
/usr/include/yaml-cpp/node/detail/node.h: In member function ‘const string& YAML::detail::node::tag() const’:
/usr/include/yaml-cpp/node/detail/node.h:33:43: error: ‘m_pRef’ was not declared in this scope
   const std::string& tag() const { return m_pRef->tag(); }
                                           ^
[... 378 lines cut ...]
                                         ^
/usr/include/yaml-cpp/node/detail/impl.h: In static member function ‘static YAML::detail::node& YAML::detail::node_data::convert_to_node(const T&, int)’:
/usr/include/yaml-cpp/node/detail/impl.h:179:10: error: base operand of ‘->’ is not a pointer
   pMemory->merge(*value.m_pMemory);
          ^
/usr/include/yaml-cpp/node/detail/impl.h:179:25: error: ‘class YAML::Node’ has no member named ‘m_pMemory’
   pMemory->merge(*value.m_pMemory);
                         ^
make[2]: *** [CMakeFiles/map_server.dir/build.make:63: CMakeFiles/map_server.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1658: CMakeFiles/map_server.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'map_server': 
[... cut ...]
==> cd /home/pi/ros_catkin_ws/build_isolated/map_server && /opt/ros/kinetic/env.sh make -j1

Command failed, exiting.
(.venv) pi@picroft:~/ros_catkin_ws $ gcc 
build_isolated/                 devel_isolated/                 kinetic-custom_ros.rosinstall   src/                            
.catkin_workspace               external_src/                   kinetic-desktop-wet.rosinstall  
(.venv) pi@picroft:~/ros_catkin_ws $ gcc -version
gcc: error: unrecognized command line option ‘-version’
gcc: fatal error: no input files
compilation terminated.

ryangswin, 2020-02-18:
Have you solved the problem? I meet the same problem when I install 'camera_calibration_parsers'. My raspbian version is Buster.

mikeferguson, 2020-03-03:
This looks like a version incompatibility - I don't think any of the Navigation folks here are RPi users, so I'd suggest posting to answers.ros.org to hopefully get some help from other RPi users.

closed 2020-03-03 by mikeferguson