ros-planning/navigation #980: move_base: make_plan service is clearing costmaps

nxdefiant, 2020-03-29:
Hi,

When using the the make_plan service of the move_base node the global costmap around the robot is cleared.

The global costmap before the service call:
[image]

The global costmap after the service call:
[image]
As one can see all obstacles close to the robot are removed.

This is probably done by the call of clearCostmapWindows() in move_base.cpp:planService().

Since my global costmap consists only of a static and inflation layer there is no way to reinsert the obstacles. I don't think that clearing the global costmap is a good idea but at least there should be a way to disable clearing.

mikeferguson, 2020-03-29:
Yeah, not sure why that clearing is there (blame says it has been there for 11 years, since the initial repo creation). I agree that this should probably be put behind a feature flag which can be toggled from a ROS param.

nxdefiant, 2020-03-30:
Or how about disabling this with a boolean passed as an input to the service?

Also later in this function there is
```
//adding the (unreachable) original goal to the end of the global plan, in case the local planner can get you there
//(the reachable goal should have been added by the global planner)
global_plan.push_back(req.goal);
```

is it also ok to disable with option?

mikeferguson, 2020-03-30:
You can't change the service definition without breaking the message API - so that can't happen in Melodic or earlier at this point. Adding ROS params (as long as their defaults don't change behavior) is acceptable in any release.

PR #981 (2020-03-31): move_base: Add options for make_plan service
Adds the following two parameters for the ~make_plan service of move_base to fix #980:

1. make_plan_clear_costmap
Whether or not to clear the global costmap on make_plan service call.

2. make_plan_add_unreachable_goal
Whether or not to add the original goal to the path if it is unreachable in the make_plan service call.
closed 2020-03-31 by mikeferguson