ros-planning/navigation #993: Map has -nan saved by Map_Saver?

RicoJia, 2020-05-09:
I ran
$ rosrun map_server map_server hw4_map.yaml 
[ INFO] [1588983232.102337577]: Loading map from image "hw4_map.pgm"
[ INFO] [1588983232.113205130]: Read a 384 X 384 map @ 0.050 m/cell

However, when I opened hw4_map.yaml, I saw
origin: [0.000000, 0.000000, -nan]

Then I tried to use map_server to load this map but failed. Any thoughts on why map_saver saved -nan in origin?

mikeferguson, 2020-05-09:
Which mapping software created the map? I presume it sent an invalid quaternion.

Regardless, almost nothing in the nav stack uses the rotation of the origin, so you can just set that -nan to zero.

RicoJia, 2020-05-09:
> Which mapping software created the map? I presume it sent an invalid quaternion.
> 
> Regardless, almost nothing in the nav stack uses the rotation of the origin, so you can just set that -nan to zero.

Oh, I had my rviz open,  and used $rosrun map_server map_saver  hw4_map.yaml

sorry I didn't specify.

mikeferguson, 2020-05-09:
You're still not answering the question - what slam package created the map you were saving? gmapping, cartographer, karto, hector_slam? Or something else?

The underlying bug is almost certainly the slam/mapping package sending an invalid quaternion, and so that's not really something we can fix here (other than to ignore it and print a warning, but that might still not yield the expected results).

closed 2020-05-13 by DLu