ros-planning/navigation #994: ROS-Noetic python3-PyKDL

Murtaza786-oss, 2020-05-25:
Just installed ROS-Noetic and ran into this bug I suspect
[image]

It seems that PyKDL.Tree() method is not working because of the segmentation fault.

mikeferguson, 2020-05-26:
So, the very little bit of PyKDL we use in navigation seems to be working (it's mostly just in test scripts). I don't think this is the right place for this ticket, unless something in navigation is not working right. I'm not sure where PyKDL errors should go, but I'd start with answers.ros.org

closed 2020-05-26 by mikeferguson

MatthijsBurgh, 2020-05-26:
@Murtaza786-oss I think this,https://github.com/foolyc/PyKDL,  is the repo of python3-PyKDL, which is completely outdated compared to the original repo, https://github.com/orocos/orocos_kinematics_dynamics

MatthijsBurgh, 2020-05-26:
[image]

Original PyKDL based on PyBind11. Sidemark, not yet released on any platform.

Murtaza786-oss, 2020-05-26:
Hello @MatthisBurgh thank you for your answer. The thing is I didn't install the repo you have mentioned above. PyKDL gets installed by default when I installed ros-noetic. and after installing I ran into this trouble. The same didn't happen with melodic

MatthijsBurgh, 2020-05-26:
I can't promise as CI runs 3.6, but 3.8 should work. It is changed compared to melodic, see https://github.com/ros/rosdistro/pull/23841/files

Murtaza786-oss, 2020-05-26:
Okay so I tried to build the repo you mentioned from source. But I land up with these errors can you please help me?

[image]

MatthijsBurgh, 2020-05-26:
@Murtaza786-oss please open an issue in the original repo