ros-planning/navigation #1003: error move_base: make_plan_clear_costmap;make_plan_add_unreachable_goal

MiPa12, 2020-06-30:
Hi,
I get the following error with the navigation package.
`catkin_ws/src/navigation/move_base/src/move_base.cpp:267:39: error: ‘class move_base::MoveBaseConfig’ has no member named ‘make_plan_clear_costmap’
     make_plan_clear_costmap_ = config.make_plan_clear_costmap;
                                       ^~~~~~~~~~~~~~~~~~~~~~~
catkin_ws/src/navigation/move_base/src/move_base.cpp:268:46: error: ‘class move_base::MoveBaseConfig’ has no member named ‘make_plan_add_unreachable_goal’
     make_plan_add_unreachable_goal_ = config.make_plan_add_unreachable_goal;
`
any tip will be appreciated, thank you

mikeferguson, 2020-06-30:
Gonna need some more info: what ROS Distro are you building under? What OS? What branch/commit are you building? Since our CI is passing on all branches, I'm guessing this is an issue with your config.

MiPa12, 2020-06-30:
Hi, I use melodic and Ubuntu 18. I cloned the melodic-devel branch

mikeferguson, 2020-06-30:
So, perhaps trying sourcing your workspace setup.bash? Maybe catkin is finding the headers installed in /opt/ros (which would not have this change, since we apparently haven't cut a new release in a while)?

pepeRossRobotics, 2020-07-01:
This is how the costmap looks like after a while, this is similar to a bug that was patched some months (or two ago)
[image]

DLu, 2020-07-03:
@pepeRossRobotics I'm unclear how your picture is related to the above compilation bug.

Tcloser, 2020-07-14:
After you catkin_make you can find  ../devel/include/move_base/MoveBaseConfig.h in your navigation package,Then
use this file to `REPLACE` the same file in /opt/ros/melodic/include/move_base/MoveBaseConfig.h

Hope it can help u~

Tcloser, 2020-07-14:
I made it , try it pls.

MiPa12, 2020-07-14:
Yes that worked, thanks.

closed 2020-07-14 by MiPa12