ros-planning/navigation #1004: map_server map_saver possible tweak

jajosheni, 2020-06-30:
Hello, this is not a but just a suggestion

here is an example yaml file

```
image: full_path/to/map.pgm
resolution: 0.1
origin: [0.0, 0.0, 0.0]
occupied_thresh: 0.65
free_thresh: 0.196
negate: 0
```

Now instead of `full_path/to/map.pgm` could you change it to be ./map.pgm as these files usually stay on the same folder?

mikeferguson, 2020-06-30:
Just drop the ./ - if you put map.pgm and it is in the same directory, it will be loaded. See examples here: https://github.com/fetchrobotics/fetch_ros/blob/melodic-devel/fetch_maps/maps/3_1_16_keepout.yaml

closed 2020-06-30 by mikeferguson

jajosheni, 2020-06-30:
I know how to make it work, as i said this is not a bug.
When saving a map instead of putting the full path I would suggest you use ./map.pgm or even map.pgm;

please read the issue then close it

thank you