ros-planning/navigation #1008: Missed inscribed radius check with footprint

pavloblindnology, 2020-07-08:
With footprint specified, the robot's center is not tested for being in `INSCRIBED_INFLATED_OBSTACLE` area
https://github.com/ros-planning/navigation/blob/b496f6891bfa8c6f471773bacb9b327e41bd36bb/base_local_planner/src/costmap_model.cpp#L63
Only footprint boundary is tested for crossing `NO_INFORMATION` and `LETHAL_OBSTACLE`.
And if all the footprints projected into the future along the considered trajectory miss `NO_INFORMATION` and `LETHAL_OBSTACLE` area, the trajectory won't be discarded by `SimpleScoredSamplingPlanner`, bcz the score returned by `CostmapModel::footprintCost` will be positive. Though the robot's center can cross `INSCRIBED_INFLATED_OBSTACLE` area, meaning it hits an obstacle.

SteveMacenski, 2020-07-08:
If the footprint is a closed shape, wouldn’t it be not possible for a lethal entry to get past that boundary with a sufficiently small dt parameter?

If we imagine a very large rectangular robot, the center cell isn’t all that interesting. I think what you’d really want to ensure no issues is to search al internal cells rather than the boundary.

DLu, 2020-07-17:
I put together a fix in the above commit, but I think this is by design. 

If you have a circular robot, then the costmap inflation makes it so that you can tell if it is hitting an obstacle with a single lookup, and so it can return the cost immediately, which includes a negative score if there is an obstacle or inscribed obstacle.

If you have a footprint, then it checks the outline for lethal obstacles. It is an optimization that doesn't check the entire interior of the footprint. It assumes that you are checking poses with enough granularity that a lethal obstacle would intersect with one of the footprints along the trajectory at some point. 

It could be optimized to also include the single lookup to immediately return if there is an obstacle, which might be the use case of what you're talking about. 

Does that make sense?

closed 2025-06-04 by mikeferguson