ros-planning/navigation #1016: catkin_make error in costmap_2d

yangyou95, 2020-07-19:
I tried to use the costmap_2d package in my workspace, when I did catkin_make, it showed errors like this:

```
Scanning dependencies of target costmap_2d_markers
[ 81%] Building CXX object navigation/costmap_2d/CMakeFiles/costmap_2d_markers.dir/src/costmap_2d_markers.cpp.o
[ 81%] Linking CXX executable /home/yang/catkin_ws/devel/lib/costmap_2d/costmap_2d_markers
/usr/bin/ld: /home/yang/ros_catkin_ws/install_isolated/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** 
```
I saw some people said this is a problem linked to the cmakelist that you need to add "tf2_ros" in the "find_package(catkin...)".
But it already in the default cmakelists in the costmap_2d package. 

Thanks for the help.

DLu, 2020-07-20:
Please specify what Ubuntu (presumably) distro you are using and what branch of costmap_2d you have checked out.

yangyou95, 2020-07-20:
> Please specify what Ubuntu (presumably) distro you are using and what branch of costmap_2d you have checked out.

Thank u, I updated my issue to give full information. I'm using DeepinV20(Based on Debian 10 buster) and melodic-devel branch.

yangyou95, 2020-07-21:
I solve this problem! When I git cloned the geometry2 package, the default version is not melodic-devel, that's why there is a problem.

Thanks anyway!

closed 2020-07-21 by yangyou95