ros-planning/navigation #1038: Discussion: Noetic API/ABI Freeze

mikeferguson, 2020-10-12:
Normally, we'd freeze the API/ABI a short time after the original release (June or so) - but since this is our last ROS release, we don't have another release to push things off to. There are a number of PRs opened that break API/ABI - and that we should probably get in.

With the MoveIt project, we've decided that API/ABI breaking changes are allowed through at least the end of the year - to give developers time to get those changes in. I'm going to propose the same for Navigation. Thoughts? @DLu @aaronhoy

aaronhoy, 2020-10-12:
This makes sense to me.

DLu, 2020-10-13:
Works for me.

mikeferguson, 2020-10-13:
I'll leave this open for comment for a week (until next Monday) - and if we don't see any compelling reason not to, I'll start making my way through a bunch of PRs.

mikeferguson, 2020-10-20:
No objections - we'll move forward with this

closed 2020-10-20 by mikeferguson