ros-planning/navigation #1044: DWAPlanner fails to compute path: 'None of the 409 first of 409 (409) points of the global plan were in the local costmap and free' and 'DWA planner failed to produce path'

ML-S, 2020-10-21:
I've already created a [answers.ros-thread](https://answers.ros.org/question/363829/dwaplanner-fails-to-get-path-with-multiple-errors/) but since until now nobody replied, I thought that maybe this is a more specific problem.

I'm trying to run a SMACH using move base flex and as controller I use the DWAPlanner. After receiving the global path (which works and also shows up in rviz), when I call the exepath-action the DWAPlanner fails to produce the path and sometimes throws the error

**"None of the 409 first of 409 (409) points of the global plan were in the local costmap and free"**

and sometimes just throws

**"DWA planner failed to produce path."**

The exact output is shown below:
[image]

Additionally, the corresponding costmap and global path looks as follows:
[image]

My parameters are set as follows:
```
planners:
  - name: 'GlobalPlanner'
    type: 'global_planner/GlobalPlanner'

controllers:
  - name: 'DWAPlannerROS'
    type: 'dwa_local_planner/DWAPlannerROS'

GlobalPlanner:
  #basic parameters
  default_tolerance: 3.0  #default is 0.0
  use_dijkstra: true  #default is true
  old_navfn_behavior: false  #default is false
  use_quadratic: true  #default is true
  use_grid_path: false  #default is false
  allow_unknown: true  #default is true
  lethal_cost: 253  #default is 253
  neutral_cost: 50  #default is 50
  cost_factor: 3  #default is 3

  #debug parameters
  publish_potential: True

DWAPlannerROS:
  max_vel_x: 3.7
  max_vel_y: 0.0
  min_vel_y: 0.0
  max_vel_trans: 3.7 
  min_vel_trans: 0.0
  max_vel_theta: 2.09 
  min_vel_theta: 0

  acc_lim_x: 2.47 
  acc_lim_y: 0
  acc_lim_theta: 4.18 

  sim_time: 2.0
  vx_samples: 3
  vy_samples: 0
  vth_samples: 20

  holonomic_robot: false 

  controller_frequency: 5.0
```

ML-S, 2020-10-22:
Reducing the size of the local costmap helps to reduce the appearance of the **"None of the 409 first of 409 (409) points of the global plan were in the local costmap and free"**-error but the DWAplanner still fails to produce the path.

ML-S, 2020-10-27:
I figured out that the issue with the **"DWA planner failed to produce path"**-warning is caused by using `track_unknown_space: true` inside the local_costmap-parameters. By setting the `track_unknown_space`-parameter to **false**, the unknown-areas are treated as free-areas resulting in the planner working perfectly fine. 

Any ideas why the DWA-planner apparently isn't able to deal with unknown-areas inside the local costmap?

mikeferguson, 2020-11-03:
Cells in a costmap that are unknown are marked as NO_INFORMATION, which, equates to a value of 255 (See cost_values.h)[https://github.com/ros-planning/navigation/blob/a9bc9c4c35a55390963db1357926ec461fcff24c/costmap_2d/include/costmap_2d/cost_values.h#L42]. A Lethal obstacle is defined as 254 - so NO_INFORMATION is worse than Lethal. AFAIK none of the default local planners have special awareness of NO_INFORMATION, so it would be treated as an obstacle.

closed 2020-11-03 by mikeferguson