ros-planning/navigation #1047: Broken API accross LTS distros: OdometryHelperRos::getRobotVel

mateus-amarante, 2020-11-04:
The argument of the method `OdometryHelperRos::getRobotVel` changed from type `tf::Stamped<tf::Pose>&` to `geometry_msgs::PoseStamped&` in the transition to melodic distro. I think the intent of this update was to remove the dependency on `tf`, however, it breaks the API.

I think all the distros should be updated to provide both versions of the method.

mikeferguson, 2020-11-04:
This was part of the port from TF to TF2. We've never really guaranteed that the API is consistent between distros - which is pretty common practice in ROS - and there were a number of API changes related to the TF2 port (for instance, the planner and costmap APIs are also pretty different because we had to change the TF buffer types and there is just no way to make it compatible between the different versions).

mateus-amarante, 2020-11-04:
It's alright. I thought they were supposed to be compatible because they share the same major version number 1.* (considering [semantic versioning](https://semver.org/) rules).

I only suggest adding `geometry_msgs::PoseStamped&` to kinetic so users can update their codes. I can make this update if you agree.

ivan1993br, 2020-11-05:
This PR would be welcome!

mikeferguson, 2022-06-20:
Kinectic is no longer supported, so all supported versions of ROS navigation now use the new interface.

closed 2022-06-20 by mikeferguson