ros-planning/navigation #1058: [kinetic] Project 'costmap_2d' specifies '/usr/include/hdf5/openmpi' as an include dir, which is not found

fmessmer, 2020-12-01:
Recently, my **Kinetic** CI jobs for packages depending on `costmap_2d` report the following failure:
```
CMake Error at /opt/ros/kinetic/share/costmap_2d/cmake/costmap_2dConfig.cmake:113 (message):
  Project 'costmap_2d' specifies '/usr/include/hdf5/openmpi' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in '${{prefix}}//usr/include/hdf5/openmpi'.  Check the website
  'http://wiki.ros.org/costmap_2d' for information and consider reporting the
  problem.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:4 (find_package)
```
I can also reproduce this locally
The latest updates that came in were:
```
The following packages were automatically installed and are no longer required:
  libhdf5-mpi-dev libhdf5-openmpi-10 libhdf5-openmpi-dev
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libproj-dev libxdmf-dev libxdmf2
The following packages have been kept back:
  librealsense2 librealsense2-dev librealsense2-dkms librealsense2-utils
The following packages will be upgraded:
  containerd.io gir1.2-poppler-0.18 libpcl-apps1.7 libpcl-common1.7 libpcl-dev libpcl-features1.7 libpcl-filters1.7 libpcl-io1.7 libpcl-kdtree1.7 libpcl-keypoints1.7 libpcl-octree1.7
  libpcl-outofcore1.7 libpcl-people1.7 libpcl-recognition1.7 libpcl-registration1.7 libpcl-sample-consensus1.7 libpcl-search1.7 libpcl-segmentation1.7 libpcl-surface1.7
  libpcl-tracking1.7 libpcl-visualization1.7 libpcl1.7 libpoppler-dev libpoppler-glib-dev libpoppler-glib8 libpoppler58 libvtk6-dev libvtk6-java libvtk6-qt-dev libvtk6.2 libvtk6.2-qt
  poppler-utils python-vtk6 slack-desktop snapd tcl-vtk6 ubuntu-core-launcher vtk6 xdg-utils
39 upgraded, 3 newly installed, 0 to remove and 4 not upgraded.
Need to get 158 MB of archives.
After this operation, 1.896 kB of additional disk space will be used.
Do you want to continue? [Y/n]
```
Apparently something changed in the recent PCL update

Any idea how to fix this?
Does `costmap_2d` not specify proper dependencies?

Although, this is a very recent issue (since latest update), this old questions seems very related:
https://answers.ros.org/question/275978/costmap_2d-expects-non-existing-include-dir/

fmessmer, 2020-12-01:
related issue is https://github.com/ros-perception/perception_pcl/issues/318

IvanVN, 2020-12-01:
I can confirm this has also started happening to me recently.

mhwasil, 2020-12-01:
I also got the same issue.

fmessmer, 2020-12-02:
@mikeferguson 
true, it turned out the issue is related to some `libvtk6` conflict which already got filed [here](https://github.com/ros/rosdistro/issues/27538) and [here](https://github.com/tfoote/vtk6-release/issues/1)
I just came here as `costmap_2d` was were the issue popped up for me....I guess we can keep this issue for visibility and close it when the `libvtk6` conflict is resolved

kyrofa, 2020-12-02:
You're right, this is related to the libvtk6 bugfix in Xenial leapfrogging the libvtk6 OR has in the ROS archive. The one in Xenial has never depended upon libhdf5-openmpi, so if you depend on it you might consider making it explicit. I'm not sure why OR's version depends on it.

mikeferguson, 2020-12-02:
We're not actually asking for anything from libhdf5-openmpi anywhere in our CMake - so it is probably trickling in from PCL.

We could add those dependencies here as a hack - but it sounds like from your comments here: https://github.com/ros/rosdistro/issues/27538 we'll still have problems until PCL gets rebuilt/rereleased

kyrofa, 2020-12-02:
Ah, not PCL, but things that depend upon PCL and libvtk6, yeah you're probably right.

fmessmer, 2020-12-17:
issue was resolved in latest sync

closed 2020-12-17 by fmessmer