ros-planning/navigation #1074: map_server not serving map on noetic

chfritz, 2020-12-21:
When I run
```
> rosrun map_server map_server map.yaml
[ INFO] [1608574452.517835560]: Loading map from image "map.png"
```
it doesn't publish the map nor offers it via a service:
```
> rosnode info /map_server_1608574509862914792 
--------------------------------------------------------------------------------
Node [/map_server_1608574509862914792]
Publications: 
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /clock [rosgraph_msgs/Clock]

Services: 
 * /map_server_1608574509862914792/get_loggers
 * /map_server_1608574509862914792/set_logger_level

contacting node http://localhost:32979/ ...
```
I've tried with three different maps, including the `testmap.yaml` (+ png) from https://github.com/ros-planning/navigation/tree/noetic-devel/map_server/test.

mikeferguson, 2020-12-23:
Can we get more info about your setup? Debs vs. source? What OS?

chfritz, 2020-12-23:
It's the vanilla ubuntu setup for noetic from debs:
```
> lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

> dpkg -la | grep map-server
ii  ros-noetic-map-server                       1.17.1-1focal.20201016.235806         amd64        map_server provides the map_server ROS Node, which offers map data as a ROS Service.

> cat /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros/ubuntu focal main

> rospack list | grep map_server
map_server /opt/ros/noetic/share/map_server
```

mikeferguson, 2020-12-23:
> It's the vanilla ubuntu setup for noetic from debs:

Hmm, so I cannot reproduce - if I fire up my navigation (https://github.com/mikeferguson/ubr_reloaded/blob/ros1/ubr1_navigation/launch/navigate.launch) I get the map no problem. I checked, I have the same exact deb installed...

chfritz, 2020-12-28:
Thanks for checking. 

I've now instrumented the map_server code and found the issue. It was stuck on this line https://github.com/ros-planning/navigation/blob/noetic-devel/map_server/src/main.cpp#L180. Once I found it I had a hunch: my gazebo simulation sets `use_sim_time` to `true` (as is typical). If you then stop gazebo `map_server` will get stuck on the above line.

I suppose this is not a bug so I'll close the ticket. Just wish it was easier to discover, since I wouldn't have guessed that statically serving a map would require a clock.

closed 2020-12-28 by chfritz

mikeferguson, 2020-12-28:
Interesting edge case - one improvement that would have made that easier to debug would be to print a message "waiting for ROS time" before the function call, and then print right after the call that the time has been received.

chfritz, 2020-12-28:
That's exactly what I was thinking. Want me to create a PR for that?