ros-planning/navigation #1093: [Bug] [Global Planner] Lethal cost is always 1 unit smaller than user setting

siferati, 2021-01-22:
Shouldn't the line below read `c < lethal_cost_ ` instead of `c < lethal_cost_ - 1`?

As it stands, a **non-lethal** cell with cost `lethal_cost_ - 1` is treated as **lethal**. This means the lethal cost is actually 1 unit smaller than the user setting (i.e. if user sets lethal cost to 100, the actual lethal cost is 99).

https://github.com/ros-planning/navigation/blob/7f22997e6804d9d7249b8a1d789bf27343b26f75/global_planner/include/global_planner/dijkstra.h#L88

If this is indeed a bug and not-intended, I can make a quick PR to fix it.

ZJUTongYang, 2021-01-28:
I think there is no bug. In the costmap (not the occupancy grid), value 254 marks only physical obstacles. Value 253 marks the C-space obstacles, i.e., the robot will definitely hit obstacles when it stays here. So it is reasonable to also avoid expanding nodes into the C-space obstacles. 

However, I noticed that your lethal cost is 100. The range of cost value changes from [0, 255] (costmap_2d) to [-1, 100] (nav_msgs/occupancyGrid), so I'm not sure whether there is a bug when you input the map in a form of occupancy grid.

siferati, 2021-01-28:
The code I linked above is working with the cost values of the costmap [0, 255] and not the occupancy grid [-1, 100], so I think we can ignore the latter for the problem at hand. The lethal_cost of 100 was merely an example I gave in the range [0, 255].

> So it is reasonable to also avoid expanding nodes into the C-space obstacles.

Yes, I agree with this. Isn't it because of this reason that the default value of lethal_cost is set to 253 and not 254?

However, if you take a look at snippet I linked above, you will notice that when using the default lethal_value 253, **the actual lethal_value would be 252 instead** - cells with cost 252 would **not** be expanded.

Basically, lethal_cost is always 1 unit smaller than the user / default setting because of the `<` sign. It should either be `<=`, or `lethal_cost_ - 1` should be replaced with `lethal_cost_`.

siferati, 2023-05-25:
>I  think the second comment here is correct - If you take a look at [cost_values.h](https://github.com/ros-planning/navigation/blob/noetic-devel/costmap_2d/include/costmap_2d/cost_values.h), "lethal - 1" is INSCRIBED_INFLATED_OBSTACLE - those are points that are also not valid (the inscribed footprint is in collision, so the cell is invalid at ANY orientation).

@mikeferguson Those values are applied to the **costmap**, not the global planner.
The `lethal_cost_` parameter of the global planner does not have the same meaning as the `costmap_2d::LETHAL_OBSTACLE` constant, that's the whole point of the parameter being dynamic reconfigurable.

And that's also why the global planner defaults this value to **253** (i.e. `costmap_2d::INSCRIBED_INFLATED_OBSTACLE`) rather than defaulting it to 254 (i.e. `costmap_2d::LETHAL_OBSTACLE`). Because for the global planner, a cell of inscribed cost is indeed a lethal cell -- the robot can not traverse it so it shouldn't plan through that cell.

https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/global_planner/cfg/GlobalPlanner.cfg#L8

However, as it stands, the global planner will also treat cells of cost 252 as lethal, which is obviously wrong.

I fixed this issue in #1240