ros-planning/navigation #1096: Costmap2d segfault

aditi741997, 2021-02-03:
Hello

I am running ros kinetic, and the costmap seg faults a lot of times, when I try to run navigation in a very big area.
Seems to be coming from the getObservation call. It'd be great if anyone can help debug this. Thanks!

Here's the gdb output: 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/home/ubuntu/catkin_ws/devel/lib/nav2d_operator/operator scan:=base_scan __name'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f7e0d7270fc in costmap_2d::Observation::Observation(costmap_2d::Observation const&) ()
   from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
[Current thread is 1 (Thread 0x7f7df5897700 (LWP 49743))]
(gdb) where
#0  0x00007f7e0d7270fc in costmap_2d::Observation::Observation(costmap_2d::Observation const&) ()
   from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#1  0x00007f7e0d7287e7 in void std::vector<costmap_2d::Observation, std::allocator<costmap_2d::Observation> >::_M_emplace_back_aux<costmap_2d::Observation const&>(costmap_2d::Observation const&) () from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#2  0x00007f7e0d7251ec in costmap_2d::ObservationBuffer::getObservations(std::vector<costmap_2d::Observation, std::allocator<costmap_2d::Observation> >&) () from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#3  0x00007f7df6743ba7 in costmap_2d::ObstacleLayer::getClearingObservations(std::vector<costmap_2d::Observation, std::allocator<costmap_2d::Observation> >&, double&) const () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#4  0x00007f7df6744543 in costmap_2d::ObstacleLayer::updateBounds(double, double, double, double*, double*, double*, double*) ()
   from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#5  0x00007f7e0d72f48b in costmap_2d::LayeredCostmap::updateMap(double, double, double) ()
   from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#6  0x00007f7e0d731cc8 in costmap_2d::Costmap2DROS::updateMap() () from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#7  0x00007f7e0d734afc in costmap_2d::Costmap2DROS::mapUpdateLoop(double) () from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#8  0x00007f7e0cb5c5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#9  0x00007f7e0e2ea6ba in start_thread (arg=0x7f7df5897700) at pthread_create.c:333
#10 0x00007f7e0da884dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

aditi741997, 2021-02-03:
It actually occurs in different places across different runs, here's a different output : 

Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/home/ubuntu/catkin_ws/devel/lib/nav2d_operator/operator scan:=base_scan __name'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007feafea69f55 in _int_free (av=0x7fead4000020, p=<optimized out>, have_lock=0) at malloc.c:4012
4012	malloc.c: No such file or directory.
[Current thread is 1 (Thread 0x7feade2a1700 (LWP 107429))]
(gdb) where
#0  0x00007feafea69f55 in _int_free (av=0x7fead4000020, p=<optimized out>, have_lock=0) at malloc.c:4012
#1  0x00007feafea6e58c in __GI___libc_free (mem=<optimized out>) at malloc.c:2975
#2  0x00007feafe78ce2b in costmap_2d::ObservationBuffer::purgeStaleObservations() () from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#3  0x00007feafe78dca6 in costmap_2d::ObservationBuffer::bufferCloud(pcl::PointCloud<pcl::PointXYZ> const&) ()
   from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#4  0x00007feafe78e605 in costmap_2d::ObservationBuffer::bufferCloud(sensor_msgs::PointCloud2_<std::allocator<void> > const&) ()
   from /home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so
#5  0x00007feadf94ca3d in costmap_2d::ObstacleLayer::laserScanCallback(boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, boost::shared_ptr<costmap_2d::ObservationBuffer> const&) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#6  0x00007feadf95ef2e in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const>) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#7  0x00007feadf97739d in message_filters::CallbackHelper1T<boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, sensor_msgs::LaserScan_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, bool) ()
   from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#8  0x00007feadf965c3f in message_filters::Signal1<sensor_msgs::LaserScan_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#9  0x00007feadf97d0dc in tf::MessageFilter<sensor_msgs::LaserScan_<std::allocator<void> > >::testMessage(ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#10 0x00007feadf97d973 in tf::MessageFilter<sensor_msgs::LaserScan_<std::allocator<void> > >::testMessages() ()
   from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#11 0x00007feadf97dc62 in tf::MessageFilter<sensor_msgs::LaserScan_<std::allocator<void> > >::add(ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#12 0x00007feadf9775c4 in message_filters::CallbackHelper1T<ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, sensor_msgs::LaserScan_<std::allocator<void> > >::call(ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, bool) ()
---Type <return> to continue, or q <return> to quit---
   from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#13 0x00007feadf95e1c7 in message_filters::Subscriber<sensor_msgs::LaserScan_<std::allocator<void> > >::cb(ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#14 0x00007feadf977945 in ros::SubscriptionCallbackHelperT<ros::MessageEvent<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) () from /home/ubuntu/catkin_ws/devel/lib/liblayers.so
#15 0x00007feafff20296 in ros::SubscriptionQueue::call() () from /home/ubuntu/catkin_ws/devel/lib/libroscpp.so
#16 0x00007feaffebf18b in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /home/ubuntu/catkin_ws/devel/lib/libroscpp.so
#17 0x00007feaffec0c63 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /home/ubuntu/catkin_ws/devel/lib/libroscpp.so
#18 0x00007feafff23769 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /home/ubuntu/catkin_ws/devel/lib/libroscpp.so
#19 0x00007feafff0589b in ros::spin() () from /home/ubuntu/catkin_ws/devel/lib/libroscpp.so
#20 0x0000000000409fce in spinner_work(ros::Publisher*) ()
#21 0x00007feaff082c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#22 0x00007feaff3536ba in start_thread (arg=0x7feade2a1700) at pthread_create.c:333
#23 0x00007feafeaf14dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

aditi741997, 2021-02-12:
More details using address sanitizer : 
==32763==ERROR: AddressSanitizer: heap-use-after-free on address 0x60d00005ee28 at pc 0x7fbf0be27935 bp 0x7fbefbd811e0 sp 0x7fbefbd80988
READ of size 12 at 0x60d00005ee28 thread T7
    #0 0x7fbf0be27934 in __asan_memcpy (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x8c934)
    #1 0x7fbf0aecc3c1 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0x11f3c1)
    #2 0x7fbf09a0ca9d in costmap_2d::Observation::Observation(costmap_2d::Observation const&) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x46a9d)
    #3 0x7fbf09a0e0a6 in void std::vector<costmap_2d::Observation, std::allocator<costmap_2d::Observation> >::_M_emplace_back_aux<costmap_2d::Observation const&>(costmap_2d::Observation const&) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x480a6)
    #4 0x7fbf09a0aaab in costmap_2d::ObservationBuffer::getObservations(std::vector<costmap_2d::Observation, std::allocator<costmap_2d::Observation> >&) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x44aab)
    #5 0x7fbefcc2f1a6 in costmap_2d::ObstacleLayer::getMarkingObservations(std::vector<costmap_2d::Observation, std::allocator<costmap_2d::Observation> >&, double&) const (/home/ubuntu/catkin_ws/devel/lib/liblayers.so+0x8d1a6)
    #6 0x7fbefcc2f3c2 in costmap_2d::ObstacleLayer::updateBounds(double, double, double, double*, double*, double*, double*) (/home/ubuntu/catkin_ws/devel/lib/liblayers.so+0x8d3c2)
    #7 0x7fbf09a14d4a in costmap_2d::LayeredCostmap::updateMap(double, double, double) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x4ed4a)
    #8 0x7fbf09a17517 in costmap_2d::Costmap2DROS::updateMap() (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x51517)
    #9 0x7fbf09a1a51b in costmap_2d::Costmap2DROS::mapUpdateLoop(double) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x5451b)
    #10 0x7fbf093875d4  (/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0+0x115d4)
    #11 0x7fbf0a5b76b9 in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76b9)
    #12 0x7fbf0a8d44dc in clone (/lib/x86_64-linux-gnu/libc.so.6+0x1074dc)

0x60d00005ee28 is located 40 bytes inside of 144-byte region [0x60d00005ee00,0x60d00005ee90)
freed by thread T0 here:
    #0 0x7fbf0be332ca in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x982ca)
    #1 0x7fbf09a09704 in costmap_2d::ObservationBuffer::purgeStaleObservations() (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x43704)

previously allocated by thread T0 here:
    #0 0x7fbf0be33602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x7fbf09a0c9dd in costmap_2d::Observation::Observation(costmap_2d::Observation const&) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x469dd)
    #2 0x7fbf09a0aec4 in costmap_2d::ObservationBuffer::bufferCloud(sensor_msgs::PointCloud2_<std::allocator<void> > const&) (/home/ubuntu/catkin_ws/devel/lib/libcostmap_2d.so+0x44ec4)
    #3 0x7fbefcc2eb4f in costmap_2d::ObstacleLayer::laserScanCallback(boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&, boost::shared_ptr<costmap_2d::ObservationBuffer> const&) (/home/ubuntu/catkin_ws/devel/lib/liblayers.so+0x8cb4f)
    #4 0x7fbefcc3f21d in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::LaserScan_<std::allocator<void> > const>) (/home/ubuntu/catkin_ws/devel/lib/liblayers.so+0x9d21d)
    #5 0x60e0000890ff  (<unknown module>)

mikeferguson, 2022-06-20:
Kinetic is no longer supported (has reached the end of life). If this persists in a more recent release, please reopen ticket with an updated trace.

closed 2022-06-20 by mikeferguson