ros-planning/navigation #1098: Resizing Costmap

HosseinSheikhi, 2021-02-11:
Hello,

I am going to add a costmap layer that gives ceiling images as input, and after segmenting images, generates a costmap. lets call this layer, `ceiling_costmap`. This `ceiling_costmap` has a bigger size than other layers, so the (minimal) layers are as follow:  
[image]

**Step 1**
The following depicts the costmap I get before applying any change to the code:
[image]

**Step2**
For the sake of efficiency, I want to just resize the `master_grid` to be in the same size as `ceiling_costmap`. 
Here is the code:
in `CeilingCostmap::UpdateCosts(...)`
[image]

and the result costmap (`plugins: ["static_layer", "ceiling_layer"]`):
[image]

**Step3**
I was playing with code to get better view, so in `static_layer.cpp`, in `processMap()`, tried to resize the `layered_costmap_` double size of `static_map`:
[image]

I expected the result costmap to be as same as the first image, except having a bigger unknown part, but here it is the costmap:
[image]

and after set a navigation goal, and updating the map:
[image]

Now here are my questions:
1- resize the `layered_costmap_` double size of `static_map` shall lead to such behavior? Is not it weird?
2- Why resizing the `mater_grid` in the second step lead to this weird costmap?

HosseinSheikhi, 2021-02-11:
The issue has been produced in the ROS2 nav stack, so moved [here](https://github.com/ros-planning/navigation2/issues/2177#issue-806150309)

mikeferguson, 2021-05-27:
Looks resolved in nav2

closed 2021-05-27 by mikeferguson