ros-planning/navigation #1105: costmap_2d　　navigation　compile　error

feng123a, 2021-03-29:
When I compile the navigation, it shows ‘noexcept’ only available with -std=c++11 or -std=gnu++11, I added set(CMAKE_CXX_FLAGS "-std=c++11") to the cmakelist, and it still does. Hope someone can tell me how to solve it.The following is the error message.

/navigation/src/navigation/costmap_2d/include/costmap_2d/layer.h:111:27: error: expected ‘;’ at end of member declaration
   inline bool isEnabled() const noexcept
                           ^
/navigation/src/navigation/costmap_2d/include/costmap_2d/layer.h:111:33: error: ‘noexcept’ does not name a type
   inline bool isEnabled() const noexcept
                                 ^
navigation/src/navigation/costmap_2d/include/costmap_2d/layer.h:111:33: note: C++11 ‘noexcept’ only available with -std=c++11 or -std=gnu++11

mikeferguson, 2021-03-29:
Need way more information:
 * What version of ROS?
 * What branch of navigation?
 * What operating system and version?

Since CI is passing on all branches, you likely are using the wrong branch for the wrong operating system / ROS version

mikeferguson, 2021-05-27:
No response - closing

closed 2021-05-27 by mikeferguson