ros-planning/navigation #1112: [AMCL] Wrong warning message when giving the wrong  value for "odom_frame_id" parameter

cailloumax, 2021-04-27:
I recently ran into an issue using AMCL, for which I was given this warning message:
```
[ WARN] [1425574394.999492201]: No laser scan received (and thus no pose updates 
have been published) for 1425574394.999407 seconds.  Verify that data is being published
on the /scan topic.
```
However, the actual issue was a miscofiguration of the _odom_frame_id_ :
`<param name="odom_frame_id"             value="/odom/ekf/imu"/>`
Here I give my topic name, in place of the frame ID required.

Would it be possible to detect the issue and give an adapted warning message ?

mikeferguson, 2021-05-27:
I think it is a bit involved - we use a tf2_ros::MessageFilter to subscribe to the laser scan - I think to differentiate between no messages and unavailable transforms you have to register a failure callback on the MessageFilter, and then set a separate timestamp in that callback and then use that timestamp to give the correct message that scans are not transformable.

Certainly a nice improvement, PRs are welcome.

Daviesss, 2023-09-17:
How was this issue resolved?  . I have similar problem . But with multi robot unable to get scan data by amcl . Robot scan topics.

[image]

Daviesss, 2023-09-18:
I solved this problem.  I had to tweak the source code . Since the ros::Duration was greater than  my laser scan interval  the warning messages pops up and my robot scan  topics will not receive scan data . So I had to make changes to that  method.

jangirrishabh, 2023-11-07:
Hello @Daviesss could you describe how you solved it?

albertrichard080, 2024-02-08:
@Daviesss Did you solved?

Daviesss, 2024-02-25:
Yes , i solved it .

closed 2025-06-04 by mikeferguson