ros-planning/navigation #1130: Costmap2DROS::getRobotFootprint() vs Costmap2DROS::getOrientedFootprint()

skpro19, 2021-06-16:
I was going through Costmap2DROS [documentation](http://docs.ros.org/en/jade/api/costmap_2d/html/classcostmap__2d_1_1Costmap2DROS.html). What is the difference between [getOrientedFootprint()](http://docs.ros.org/en/jade/api/costmap_2d/html/classcostmap__2d_1_1Costmap2DROS.html#a8f4eafac38f214509cd00d1541ec8a84) and [getRobotFootprint()](http://docs.ros.org/en/jade/api/costmap_2d/html/classcostmap__2d_1_1Costmap2DROS.html#add1c9e06ffbae7fecc956c3c377f869e) methods?

mikeferguson, 2021-06-17:
The oriented footprint is just the robot footprint relocated to new pose.

closed 2021-06-17 by mikeferguson

skpro19, 2021-06-17:
You mean that the `getOrientedFootprint` function takes in argument the `desired pose` of the bot (in the `world` frame) and outputs the footprint co-ordinates in the `world` frame?