ros-planning/navigation #1145: fatal error: ros/package.h: No such file or directory

wrmichaelmak, 2021-07-22:
I had this issue while building `navigation` from source using `catkin build`. I had fixed it but thought that it will be good to put out the fix for it. I edited the `CMakLists.txt` and added `roslib` under `find_package()` and `catkin_package(... CATKIN_DEPEND ... roslib)`

DLu, 2021-08-19:
Can you please provide the specific error that you saw and what version you encountered it.

wrmichaelmak, 2021-08-19:
Hi, the specific error I faced is exactly as stated on the issue title. I encountered the problem while using ROS Noetic

DLu, 2021-08-19:
But on which package?

wrmichaelmak, 2021-08-19:
Sorry for the lack of information, I was having problems with the map_server package

mikeferguson, 2022-06-20:
Seeing as all branches are passing CI without this extra dependency, I'm thinking this is a local configuration issue. If the problem persists, please post to answers.ros.org

closed 2022-06-20 by mikeferguson

alfanrizaldy, 2022-12-14:
Hello, how you solve this problem? @wrmichaelmak i'm facing this problem too and still can't found the answer
thank you