ros-planning/navigation #1148: Request for Tutorials for DWA Planner in ROS Wiki

argahsuknesib, 2021-07-26:
Good day,
I think it will be a great idea to include some tutorials, same as TEB planner's tutorials.
Thank you.

DLu, 2021-07-27:
We welcome contributions for this feature. 

For reference: http://wiki.ros.org/teb_local_planner/Tutorials

SANTHOSH-MAMIDISETTI, 2023-06-09:
@argahsuknesib @DLu  , I would love contributing to opensource and I am planning to build my career full time in robotics and AI . Can I work over this ? or ,i s this issue solved , else if there is some other issue you would want me to work on , please do assign me .

argahsuknesib, 2023-06-09:
Hello @SANTHOSH-MAMIDISETTI, I am not working with ROS anymore. Please feel free to work on this. I guess @mikeferguson can assign you to the issue?

SANTHOSH-MAMIDISETTI, 2023-06-09:
> Hello @SANTHOSH-MAMIDISETTI, I am not working with ROS anymore. Please feel free to work on this. I guess @mikeferguson can assign you to the issue?

Thank you @argahsuknesib  for your quick reply , Can you , if possible please brief me / describe what did you work on , what were your plans to this or something like that as that would help me a lot in understanding this and even in implementing too . with the fact that I am a beginner , that would definitely help a lot , Thanks!

SANTHOSH-MAMIDISETTI, 2023-06-09:
> Hello @SANTHOSH-MAMIDISETTI, I am not working with ROS anymore. Please feel free to work on this. I guess @mikeferguson can assign you to the issue?

@mikeferguson Can I be assigned this (and maybe even briefed a bit!, like what kind of tutorials are needed , if anything in specific ) ?  thanks!

closed 2025-06-04 by mikeferguson