ros-planning/navigation #1150: Version without ros

oym1994, 2021-07-27:
Hi, dear developer,

I have a wearied problem. My supervise don't like ROS and it's impossible to convince him. So have u also maintained a non-ROS version?  or some parts without ROS, such as non-ROS costmap.
Btw, if there is no non-ROS version,  can u give some advise on removing ROS  from the project?

Thank you for your attention!

reinzor, 2021-07-27:
I am afraid If you would like to use these packages without ROS you are going to have a hard time. However, you could use some internal libraries without the ROS communication dependencies, e.g. costmap-2d or specific planner implementations. If you are not interested in using the ros communication libs and/or the build environment, I would suggest to go for alternative software packages.

mikeferguson, 2021-07-28:
This is the ROS Navigation stack, it would be an epic effort to remove the ROS from it.

closed 2021-07-28 by mikeferguson