ros-planning/navigation #1166: The intention of stuck_left & stuck_right in base_local_planner

ban-masa, 2021-08-31:
I want to know the intention of stuck_left & stuck_right in base_local_planner.
I think stuck_left means that the robot has been stuck during left rotation and prevents rotating left.
However, the following lines mean that left rotation(=thetav > 0) will be selected if stuck_right is false.
https://github.com/ros-planning/navigation/blob/2b807bd312fac1b476851800c84cb962559cbc53/base_local_planner/src/trajectory_planner.cpp#L679-L692
In addition, I can't understand why stuck_left is set true when rotating_left is true.
https://github.com/ros-planning/navigation/blob/2b807bd312fac1b476851800c84cb962559cbc53/base_local_planner/src/trajectory_planner.cpp#L704-L714
At least, the above code contradicts the following code.
https://github.com/ros-planning/navigation/blob/2b807bd312fac1b476851800c84cb962559cbc53/base_local_planner/src/trajectory_planner.cpp#L801-L811

I think the implementation of stuck_left & stuck_right is wrong.
Does anyone know the intention of this implementation?

mikeferguson, 2022-06-20:
Ok, so here's my understanding:

 * Initially all rotating/stuck are set to FALSE.
 * Anytime we travel oscillation_reset_dist the rotating/stuck are again set to FALSE.
 * Eventually we do an in-place rotation - let's say it is to the left (positive theta)
   * The first time through, that left rotation is selected as best_traj, and then we set rotating_left flag to true
   * The second time through (this gets called at 10-20hz on most robots), we set stuck_left to true (since rotating_left was true).
   * Every time after the second time through the loop, we can only accept left rotations as the best_traj (since right rotations imply that vtheta < 0 and stuck_left is now true). This will make sure that we don't oscillate between left and right commands - we'll just keep rotating left now until either we get a forward movement, or we trip the oscillation timeout (which is really a "lack of linear progress" timeout).

This whole strategy could probably certainly be improved - but I think the logic is correct for the simplistic approach (it was originally not correct - [and was fixed](https://github.com/ros-planning/navigation/commit/fd0177b18088091fdee6df3806aa5b5489a075c0) along with a whole slew of other bugs in the April 2015 time when I was trying to make everything run well for the original Fetch Research Manipulator release)

mikeferguson, 2023-02-15:
Closing due to inactivity

closed 2023-02-15 by mikeferguson