ros-planning/navigation #1188: ros-melodic-costmap-2d dependancy error

lianghongzhuo, 2022-01-26:
I got the follwing error when doing `sudo apt upgrade`
system: ubuntu18.04 ros melodic
```
The following packages have unmet dependencies:
 ros-melodic-costmap-2d : Depends: ros-Melodic-message-filters but it is not installable
```
I guess there is a typo in the dependancy? Where  `ros-Melodic-message-filters` has a capital "M"

mikeferguson, 2022-01-26:
I don't see how that could have happened - A) we haven't had a release in a while, and B) the "ros-melodic-" part gets added by bloom (so it is automated when we do the release)

lianghongzhuo, 2022-01-26:
I also confused how this could happen.

mikeferguson, 2022-01-26:
So.. I just downloaded http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-costmap-2d/ros-melodic-costmap-2d_1.16.7-1bionic.20210921.222823_amd64.deb and extracted the control file: 

```
Package: ros-melodic-costmap-2d
Version: 1.16.7-1bionic.20210921.222823
Architecture: amd64
Maintainer: David V. Lu!! <davidvlu@gmail.com>
Installed-Size: 2261
Depends: libboost-filesystem1.65.1, libboost-system1.65.1, libboost-thread1.65.1, libc6 (>= 2.14), libconsole-bridge0.4, libgcc1 (>= 1:3.0), libstdc++6 (>= 5.2), libtinyxml2-6 (>= 5.0.0), ros-melodic-dynamic-reconfigure, ros-melodic-geometry-msgs, ros-melodic-laser-geometry, ros-melodic-map-msgs, ros-melodic-message-filters, ros-melodic-message-runtime, ros-melodic-nav-msgs, ros-melodic-pluginlib, ros-melodic-rosconsole, ros-melodic-roscpp, ros-melodic-rostest, ros-melodic-sensor-msgs, ros-melodic-std-msgs, ros-melodic-tf2, ros-melodic-tf2-ros, ros-melodic-visualization-msgs, ros-melodic-voxel-grid
Section: misc
Priority: optional
Homepage: http://wiki.ros.org/costmap_2d
Description: This package provides an implementation of a 2D costmap that takes in sensor data from the world, builds a 2D or 3D occupancy grid of the data (depending on whether a voxel based implementation is used), and inflates costs in a 2D costmap based on the occupancy grid and a user specified inflation radius.
 This package also provides support for map_server based initialization of a costmap, rolling window based costmaps, and parameter based subscription to and configuration of sensor topics.
```

I don't see any problem here

lianghongzhuo, 2022-01-26:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  ros-melodic-costmap-2d ros-melodic-dynamic-reconfigure
The following packages will be upgraded:
  ros-melodic-costmap-2d ros-melodic-dynamic-reconfigure
2 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

```
Just run upgrade again, seems someone fixed it silently?

tfoote, 2022-03-03:
It sounds like you might have gotten an inconsistent state in your apt-update. As it's working and there's nothing to track to follow up I'm going to close this.

closed 2022-03-03 by tfoote