ros-planning/navigation #1195: DWA can not add parameter

LGK199408, 2022-04-11:
I had added a parameter into DWAPlanner.cfg
`gen.add("target_occupied", double_t, 0, "when target was occupied there is another xy_goal_tolerance", 0.5, 0.0, 1.0)
`
and then  invoked the parameter in dwa_planner_ros.cpp
`double target_occupied_xy1 = config.target_occupied;`

and then catkin_make 
error was occured...

`error: ‘class dwa_local_planner::DWAPlannerConfig’ has no member named ‘target_occupied’
`

DLu, 2022-04-11:
Does the error persist if you try building again?

LGK199408, 2022-04-12:
Yes @DLu

LGK199408, 2022-04-12:
[image]
[image]

mikeferguson, 2022-06-20:
This appears to be some issue with your local workspace - not a bug in the navigation stack (this repo). Please post to answers.ros.org as there are many more folks over there to help out with debugging.

closed 2022-06-20 by mikeferguson