ros-planning/navigation #1202: Navigation Issues between Kinetic and Noetic

CycleMark, 2022-06-19:
Hi,
   I posted this issue on rosanswers (`https://answers.ros.org/question/397740/navigation-issues-between-kinetic-and-noetic/`) - but not luck getting an answer. So wanted to see if anyone here can help specifically. 

Put simply upgraded ROS from Kinetic (where all was working fine) to Noetic and have encountered navigation issues. Mainly around the robot gittering/dithering when going through the likes of a door way. Usual ends up in several attempts with cost clearing and recovery modes entered. Wasn't an issue at all in Kinetic. Robot navigated around without issue. Straight through doorways countless times without incident. 

I know there will have been a number of changes between these releases and where parameters have changed I've accounted for them (unless these parameters now mean something different?). I've created a side by side here - `https://www.dropbox.com/s/10997q4w7bzj2ut/ROS%20Mapping%20issues%20analysis.xlsx?dl=0` but can't see anything that's differnt per sa.

Not sure if the problem(s) lie in the path planning, map (global or local) of some combination.

Everything on the hardware front remains the same.

Advice would be much appreciated as I've got a platform at the moment that's non-functional.

Thanks and Regards

Mark

mikeferguson, 2022-06-19:
I don't think anyone is going to be able to debug this for you from the provided information - specifically:

> Not sure if the problem(s) lie in the path planning, map (global or local) of some combination.

I'd suggest starting by visualizing each of these - look at the global plan - is it good? If so move onto visualizing the local plan. And if that looks correct, check what the cmd_vel output looks like (versus your odometry as well to make sure it's not actually an issue with something below the navigation stack - even though it's the same hardware, there may have been updates to lower level drivers when jumping forward 4+ years in software)

CycleMark, 2022-06-19:
Hi Mike,

Thanks - yeah I understand the changes will be quite a few over that period of time. 

I'd already looked at the plans and maps (local and global) in Rviz vs Kinetic and they did look the same. I usual come to these forums when I've run out of ideas myself. Just in case there's a "X changed specifically that would cause Y" type scenario that someone knows about.

I've started diffing the code - I can see there are some changes around the recovery settings but not yet managed to see why (or indeed if) they're affecting the current performance - or lack therefore.

mikeferguson, 2022-09-22:
Closing due to lack of activity - if the issue can be further nailed down, just reopen

closed 2022-09-22 by mikeferguson