ros-planning/navigation #1205: Costmap_2d : Local Costmap Freezes

charvi-077, 2022-06-30:
Hi  all, 

Ubuntu : 18.04 and 20.04 (testing on both)
Ros version : ROS1 Melodic and Noetic (testing on both)

I am working on generating the local costmap (rolling window) using 2d lidar data. For generating it I am only using oblacle layer plugin and inflation layer plugin. I have make sure to configure the tf properly for all the 2d lidars. But still at some time the costmap get freezes as shown in below gif. 

[image]

As shown in gif above, the lower right pixels and even in middle pixels got freezed and are not updating. In gif the coloured data is lidar data, and black is costmap data obstacle + inflation to differentiate between realtime costmap data and freezed data in costmap. Moreover, sometimes the full costmap gets freezes. 

So, I am unable to resolve the above issues, and would really appreciate if someone could suggest how to resolve it or related pointer to this issue.

sun421-cloud, 2023-08-29:
hello, i also encounter this question, did you work out it?

charvi-077, 2023-08-29:
Hi @sun421-cloud , 
No its not resolved yet, but yeah I just did a workaround to rather reset the local costmap on every incoming 2d lidar scan data, so here I changed the obstacle layer plugin to instead reset the the costmap instead of raytracing free space. Note, this was just wrt our application of local costmaps and yeah this won't work for global costmaps ( for marking and clearing obstacle information globally).

sun421-cloud, 2023-08-30:
Thank you very much for your answer, and may I know that you always clear the whole local costmap before updating every incoming 2d lidar scan data? Are your lidar datas 360 degrees? no shielding Angle?
In addition, I would like to ask how you adjust the shape of the local costmap? I'm currently using a square, and notice you're using a rectangle.

AngeLoGa, 2023-11-08:
> Thank you very much for your answer, and may I know that you always clear the whole local costmap before updating every incoming 2d lidar scan data? Are your lidar datas 360 degrees? no shielding [...]

This can be done using the parameters in the costmap2d package, please see the wiki ([see here](http://wiki.ros.org/costmap_2d))

sun421-cloud, 2023-12-21:
> This can be done using the parameters in the costmap2d package, please see the wiki ([see here](http://wiki.ros.org/costmap_2d))

ok, i get it, thanks!