ros-planning/navigation #1210: catkin make could NOT find tf2_sensor_msgs to work with navigation package

mehdikerbelae, 2022-08-22:
Could NOT find tf2_sensor_msgs (missing: tf2_sensor_msgs_DIR)
-- Could not find the required component 'tf2_sensor_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf2_sensor_msgs"
  with any of the following names:

    tf2_sensor_msgsConfig.cmake
    tf2_sensor_msgs-config.cmake

  Add the installation prefix of "tf2_sensor_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_sensor_msgs_DIR" to a directory containing one of the above files.
  If "tf2_sensor_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  navigation/costmap_2d/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/gis/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/gis/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:1860: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

melodic, ubuntu 18

mikeferguson, 2022-08-22:
The dependency is properly specified in the package.xml (otherwise our CI would not pass). You need to install required dependencies when building from source - the easiest approach is to use rosdep - from the root of your workspace you want to run "rosdep install --from-paths src --ignore-src --rosdistro=melodic -y".

mehdikerbelae, 2022-08-22:
rosdep install --from-paths ~/catkin_ws/src --ignore-src --rosdistro=melodic -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
amcl: [python3-pykdl] defined as "not available" for OS version [bionic]

SteveMacenski, 2022-08-22:
Ask on ROS Answers.

closed 2022-08-22 by SteveMacenski

mehdikerbelae, 2022-08-22:
thanks a lot, I did.

mikeferguson, 2022-08-22:
I think you have the wrong branch checked out - make sure you're using melodic-devel (noetic-devel won't work on 18.04)