ros-planning/navigation #1233: MultiThreadedSpinner move base

macstepien, 2023-01-23:
Hi,
I noticed that there is a commented out definition of `MultiThreadedSpinner` in the `move_base_node.cpp`. I'm wondering if there is any particular reason that it is commented out and if you are aware of any risks that come with using MultiThreadedSpinner in move_base?

To give a bit more context, in our application map can get quite large and I noticed that when the global costmap's inflation layer updates, it somehow blocks the local costmap's sensor callbacks (ObstacleLayer with LaserScan and PointCloud2) . As observation buffers are not updated, it results in an emergency stop of the robot. I noticed that using even two threads for spinning solves the problem. I'm not yet sure what exactly can block the callback queue - I haven't found anything suspicious in the inflation layer (we also have some custom costmap layers, but it seems to happen during the inflation layer update).
Thanks,
Mac

mikeferguson, 2023-02-15:
move_base is already multi threaded in a few spots - the planner spins up a separate planning thread and each costmap has it's own update thread. I think adding additional threads for the main loop probably has some risk since it is likely there are things touching the same data that are not appropriately locked.

mikeferguson, 2023-02-15:
Here is an example: https://github.com/ros-planning/navigation/issues/1173

macstepien, 2023-03-01:
Understood, thanks for your response!

closed 2023-03-01 by macstepien