ros-planning/navigation #1250: Amcl not working properly .

Daviesss, 2023-09-17:
I have a problem . Which i have been on for the past one week. I dont know why this pops up . I am working on Multi robot collaboration(SLAM) . But for some reasons the error below pops  up anytime i use Amcl (the robot cant be able to localize on the map due to this error) . I will be glad if i get a help .

[ WARN] [1694953982.886336265, 170.233000000]: No laser scan received (and thus no pose updates have been published) for 170.233000 seconds.  Verify that data is being published on the /robot3/scan topic.

[image]

Daviesss, 2023-09-18:
I solved this problem. I had to tweak the source code . Since the ros::Duration was greater than my laser scan interval the warning messages pops up and my robot scan topics will not receive scan data . So I had to make changes to that method.

Bankbuster777, 2024-01-11:
> I solved this problem. I had to tweak the source code . Since the ros::Duration was greater than my laser scan interval the warning messages pops up and my robot scan topics will not receive scan data . So I had to make changes to that method.

did you mean this duration? 
https://github.com/ros-planning/navigation/blob/9ad644198e132d0e950579a3bc72c29da46e60b0/amcl/src/amcl_node.cpp#L509C47-L509C47

Daviesss, 2024-02-25:
To reduce the number of TF_REPEATED_DATA warnings the laser scan publish rate needs to be reduced, because amcl depends on rate of incoming laser messages. I edited the source code of amcl to suit the coordination. Works fine now .

mateusguilhermedasilva, 2024-07-27:
> I solved this problem. I had to tweak the source code . Since the ros::Duration was greater than my laser scan interval the warning messages pops up and my robot scan topics will not receive scan data . So I had to make changes to that method.

I have the same error message:

"[WARN] [1722117737.201293908]: No laser scan received (and thus no pose updates have been published) for 1722117737.201188 seconds. Verify that data is being published on the /scan topic."

I can see that there are messages being published in the /scan topic and that amcl is subscribing to this topic.

How did you go about solving it?

mikeferguson, 2025-06-04:
ROS 1 is now EOL. Please consider migrating to https://github.com/ros-navigation/navigation2

closed 2025-06-04 by mikeferguson