ros-planning/navigation #1255: Possibility of remapping tf publication of AMCL

AngeLoGa, 2023-11-08:
I'm trying to perform Localization in outdoor and indoor situations, and I would like to remap the /tf publisher from AMCL node so I can redirect it to a mux where I could select which algorithm transform should be used.

If I'm not wrong, using the remap tag would also affect not the publisher but also the subscriber, and I can't do that since AMCL needs the transform from the laser_frame to odom.

I thought this idea because I haven't seen any way to stop AMCL from publishing, without exiting the node.

Could I do this somehow? Or am I doing something wrong?

mikeferguson, 2023-11-08:
There is a parameter "tf_broadcast" that defaults to true - but if you set it to false, AMCL will no longer publish the TF data, just the amcl_pose topic (which includes the pose with covariance). Your "mux" would then subscribe to the amcl_pose topic and use that data to create the proper TF data to publish.

AngeLoGa, 2023-11-09:
That seems the perfect way to do it. I thought AMCL main focus was to publish the tf, but didn't remember that it still works without publishing it. Thanks!

closed 2023-11-09 by AngeLoGa