ros-planning/navigation #1257: The "global_planner/GlobalPlanner" plans paths through obstacles and unknown areas occasionally

ahmedadamji, 2023-12-18:
I am encountering an intermittent issue while using the ROS navigation stack with global_planner/GlobalPlanner as the base global planner. Occasionally, the planned path generated by the global planner incorrectly goes straight through walls represented in the static map. This issue does not occur consistently but happens sporadically, which leads me to believe there might be an underlying problem with how the planner is being used or configured.

Environment:

ROS Version: ROS Noetic
Operating System: Ubuntu 20.04

Configuration Details: 

Relevant parameters set for global_planner/GlobalPlanner.
base_global_planner: "global_planner/GlobalPlanner" #"navfn/NavfnROS"
GlobalPlanner:
  default_tolerance: 0.1
  allow_unknown: false
  visualize_potential: false
  publish_potential: false
  use_dijkstra: true
  orientation_mode: 1
  orientation_window_size: 5
  unknown_cost_value: 255
  outline_map: true 

Costmap configuration:
global_costmap:
  global_frame: map
  robot_base_frame: base_link
  update_frequency: 4.0
  publish_frequency: 0.5
  static_map: true
  resolution: 0.1
  map_topic: "/map"
  width: 100
  height: 100

  transform_tolerance: 0.5
  plugins:
    - {name: static_layer,               type: "costmap_2d::StaticLayer"}
    - {name: image_seg_obstacle_layer,   type: "costmap_2d::ObstacleLayer"}
    - {name: inflation_layer,            type: "costmap_2d::InflationLayer"}
 
static_layer:
  enabled:              true
  map_topic:            "/map"
  unknown_cost_value:   255
  track_unknown_space:  true
  trinary_costmap:      true

Expected Behavior:
The global planner should consistently avoid planning paths through walls or obstacles as defined in the static map.

Actual Behavior:
Intermittently, the planner generates paths that go straight through walls.

Nisarg236, 2023-12-18:
Can you share a screenshot or the map image with start and goal location. It will help debug the issue.

MortezaHagh, 2024-01-10:
I experienced similar issue which I described here:
https://github.com/ros-planning/navigation/issues/1254#issue-1957220409

mikeferguson, 2025-06-04:
ROS 1 is now EOL. Please consider migrating to https://github.com/ros-navigation/navigation2

closed 2025-06-04 by mikeferguson